analogdevicesinc/scopy #1050: <Instrument>: <Title>

jpnielsen, 2021-03-15:
Debian 11 (Bullseye)

uname -a
Linux debian 5.10.0-4-amd64 #1 SMP Debian 5.10.19-1 (2021-03-02) x86_64 GNU/Linux

KDE plasma desktop
flatpak        1.10.1-4 
```

flatpak run org.adi.Scopy

connect to device
click "oscilloscope", click "Run"
click "autoset"
program exits

/var/log/messages

Mar 15 19:02:21 debian kernel: [10668.242378] traps: stream_to_vec15[39601] trap invalid opcode ip:7f2d099d4a3b sp:7f2c6bffe5c8 error:0 in libgmp.so.10.4.0[7f2d099b5000+59000]

flatpak run org.adi.Scopy

connect to device
start "Spectrum Analyzer" -
click "Run"
program exits

/var/log/messages 

Mar 15 19:08:44 debian kernel: [11051.163849] traps: stream_to_vect7[40154] trap invalid opcode ip:7fc24a6aca3b sp:7fc1c6ffc5c8 error:0 in libgmp.so.10.4.0[7fc24a68d000+59000]
Mar 15 19:08:44 debian kernel: [11051.163862] traps: stream_to_vect8[40137] trap invalid opcode ip:7fc24a6aca3b sp:7fc1e57f95c8 error:0
```

```
dpkg -l | grep libgmp
ii  libgmp-dev:amd64                              2:6.2.1+dfsg-1                 amd64        Multiprecision arithmetic library developers tools
ii  libgmp10:amd64                                2:6.2.1+dfsg-1                 amd64        Multiprecision arithmetic library
ii  libgmpxx4ldbl:amd64                           2:6.2.1+dfsg-1                 amd64        Multiprecision arithmetic library (C++ bindings)

```

```
flatpak list
Name                              Application ID                               Version          Branch         Origin               Installation
Scopy                             org.adi.Scopy                                                 master         scopy-origin         system
default                           org.freedesktop.Platform.GL.default                           19.08          flathub              system
openh264                          org.freedesktop.Platform.openh264            2.1.0            2.0            flathub              system
Breeze Gtk theme                  org.gtk.Gtk3theme.Breeze                                      3.22           flathub              system
KDE Application Platform          org.kde.Platform                                              5.14           flathub              system

```

adisuciu, 2021-03-16:
Are you using a VM for this ?
flatpak bundles all the libraries needed inside. At build time, libgmp is built from source. There may be some configuration incompatibilities between the build machine and your target machine, although we haven't seen much of this. I'll try to rebuild the flatpak with --enable-fat parameter for libgmp- https://gmplib.org/manual/Build-Options . Can you give it a try - when it's done https://github.com/analogdevicesinc/scopy/runs/2096315181 ? 
-Adrian

jpnielsen, 2021-03-16:
No - I'm running on hardware. 

AMD fx-8350 CPU, ASrock Pro3 motherboard, 24GB RAM
Radeon RX580 pci-e gfx
Kingston SSD

I have other PC's too - but haven't tried there yet.

jpnielsen, 2021-03-16:
The new flatpak from:
https://github.com/analogdevicesinc/scopy/suites/2244955973/artifacts/47401710

Solved the two described problems.

Thank you.

jpnielsen, 2021-03-16:
I do get some errors on the console, but the program doesn't crash like before.

ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20210316 18:24:14.890820  1698 buffer.cpp:479] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.

adisuciu, 2021-03-16:
This kind of output is expected. we will tidy up the application output with a future update. But if it doesn't crash I'd close this for now.

closed 2021-03-16 by jpnielsen

adisuciu, 2021-03-16:
Just a followup question .. the flatpak that crashed before .. did you download that from a nightly or a release ?

jpnielsen, 2021-03-16:
It was the latest release - or atleast 99% sure it was.

adisuciu, 2021-03-16:
Alright, thanks ! So it might not have been the --enable-fat in the libgmp build but I think it's a better build recipe for our usecase either way. Thanks again.

jpnielsen, 2021-03-16:
I have not downloaded the nightly builds for a while.

When i look at https://ci.appveyor.com/project/analogdevicesinc/scopy - I cannot find the flatpak builds that didn't fail. Most builds seems to be either failed or cancelled.

adisuciu, 2021-03-16:
This one should not have the --enable-fat parameter if you want to give it a try. Then we'll know whether the build parameter fixed it for you or not. https://github.com/analogdevicesinc/scopy/runs/2093385633

jpnielsen, 2021-03-16:
https://github.com/analogdevicesinc/scopy/runs/2093385633

This version exits like before, when i do "Autoset" - or "Spectrum
Analyzer" -> "Run"

And I'm running KUbuntu 20.04.2.0 now.

Mar 16 20:21:54 desktop kernel: [ 3724.023066] traps:
stream_to_vect8[38386] trap invalid opcode ip:7f7f2c314
6fb sp:7f7e33ff6d28 error:0 in libgmp.so.10.4.0[7f7f2c2f5000+59000]
Mar 16 20:21:54 desktop systemd[924]: flatpak-org.adi.Scopy-38222.scope:
Succeeded.

adisuciu, 2021-03-16:
Awesome. So the build parameter fixed it. Thanks a ton !