analogdevicesinc/scopy #1060: hello unable to build scopy on windows msys2

keleisibo, 2021-03-23:
build the project like this : https://wiki.analog.com/university/tools/m2k/scopy/build-windows
[image]

and occur the error below:
FAILED: CMakeFiles/scopy.dir/resources/properties.rc.res
CreateProcess failed: The system cannot find the file specified.
[image]
help me!

adisuciu, 2021-03-24:
The Windows build instructions are not great. Even on our side, things go sideways sometimes. This is why we use the CI to build the latest version which is somewhat stable. We will create new build scripts in the future. Why do you want to build Scopy on Windows. You can get the latest from our CI. Currently this is the latest working build - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38266090

keleisibo, 2021-03-24:
you mean that I execute the script "build_appveyor_mingw.sh"，and then build success?

keleisibo, 2021-03-24:
Thank you for answering my question，you mean that I execute the script "build_appveyor_mingw.sh"，and then build success?

adisuciu, 2021-03-24:
No, you need to install gnuradio, then scopy-mingw-build-deps , then build scopy. The scripts do a good job of building this in the cloud (appveyor), but don't work that well on users machines without tweaking. We primarily use Linux to develop(which make things a lot easier) and only once in a while use Windows to test. 
Like I said, your best bet is either compiling on Linux if you want to do development or using already built binaries for the platforms we provide (win/linux/mac). We will create some scripts that are easier on user's machines on Windows, but meanwhile I'd stick to the things that work - building on linux/using the provided installers (for Windows)

keleisibo, 2021-03-25:
got it,Thank you

closed 2021-03-25 by adisuciu