analogdevicesinc/scopy #1063: Logic analyzer: Not going low correctly

kes-electronics, 2021-03-24:
**Environment:**
 - OS: Windows 10 build 1902
 - FW Version 0.26
 - SW Version 1.2.0

**Describe the bug**
Logic analyzer trace will only show a low signal when followed by a high signal. The digital IO shows correct level.

**To Reproduce**
Steps to reproduce the behaviour:
1. Build switch circuit, using 5V supply from ADALM2000 (issue also present with external supply). Connect output to DIO0.
[image]
2. Use logic analyzer, 20 ksps ,200 ksamples, enable DIO0, rising edge trigger.
3. Run single, press switch to trigger analyzer
4. See results

**Expected behavior**
Trace to match switch presses (and digital IO)

**Screenshots**
[image]

**Additional context**
Work computer, cannot find .ini file!
Similar issue when I use one digital IO pin as output connected to another set as input, and take trace on logic analyzer as output state is toggled.

adisuciu, 2021-03-24:
Yes, this issue should be fixed in the latest version. You can download it here - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38266090 . We will create a release candidate in the following days.

kes-electronics, 2021-03-24:
Excellent, thank you.

closed 2021-05-05 by adisuciu