analogdevicesinc/scopy #1071: <Instrument>: <Short description of the bug>

yazzy18, 2021-03-31:
OS Version: macOS 10.16
FW Version: 
SW Version: e02cebe

Description of the bug:<description>
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: /Users/yasminabumaye/.config/analog.com/Scopy.ini
Please consider attaching it.

teoperisanu, 2021-03-31:
Can you give us more information, such as the bug description and the steps to reproduce it. Also there is a new Scopy version: https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc1.

yazzy18, 2021-03-31:
I don’t exactly know what the bug was since i exited out of the report very
quickly on accident. I’m so sorry I can’t be of much help, but if I get
another crash report, I’ll make sure to document big description. As for
the steps I took to get to that bug, it was not intentional haha. I believe
I downloaded the newest version of Scopy but I’m not sure if it’s
compatible with OS Big Sur.

adisuciu, 2021-03-31:
Closing this .. reopen after trying v1.3.0-rc, it reproduces and you have some reproduction steps.

closed 2021-03-31 by adisuciu

clarsen, 2021-09-07:
MacOS 11.5.2
Using Scopy v1.3.0 (june 14 release).

Pluto plugged in directly to macbook (not via hub).  Powered from USB-C, not via external power.

Crashes on attempting to connect to Pluto.

details from problem report
```
Process:               Scopy [77683]
Path:                  /Users/USER/*/Scopy.app/Contents/MacOS/Scopy
Identifier:            org.adi.Scopy
Version:               ??? (1.3.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Scopy [77683]
User ID:               501

Date/Time:             2021-09-07 16:36:58.067 -0700
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        68588E0E-1F84-4E7C-DC20-96C54D81251C

Sleep/Wake UUID:       961160BF-3371-4813-88A9-A0B24F8568DA

Time Awake Since Boot: 910000 seconds
Time Since Wake:       770 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [77683]

VM Regions Near 0:
--> 
    __TEXT                      10595e000-105e4a000    [ 5040K] r-x/r-x SM=COW  /Users/*/Scopy.app/Contents/MacOS/Scopy

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.adi.Scopy                 	0x0000000105befc98 0x10595e000 + 2694296
1   org.adi.Scopy                 	0x0000000105bef64e 0x10595e000 + 2692686
2   org.qt-project.QtCore         	0x0000000107931182 0x10771a000 + 2191746
3   org.qt-project.QtWidgets      	0x0000000106c5ae2f 0x106b60000 + 1027631
4   org.qt-project.QtWidgets      	0x0000000106c5aca9 0x106b60000 + 1027241
5   org.qt-project.QtWidgets      	0x0000000106c5be6c QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 268
6   org.qt-project.QtWidgets      	0x0000000106bacd0a QWidget::event(QEvent*) + 442
7   org.qt-project.QtWidgets      	0x0000000106b73a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
8   org.qt-project.QtWidgets      	0x0000000106b76944 QApplication::notify(QObject*, QEvent*) + 7444
9   org.qt-project.QtCore         	0x00000001079002b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
10  org.qt-project.QtWidgets      	0x0000000106b74362 QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 898
11  org.qt-project.QtWidgets      	0x0000000106bcbca5 0x106b60000 + 441509
12  org.qt-project.QtWidgets      	0x0000000106bca3de 0x106b60000 + 435166
13  org.qt-project.QtWidgets      	0x0000000106b73a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
14  org.qt-project.QtWidgets      	0x0000000106b74e02 QApplication::notify(QObject*, QEvent*) + 466
15  org.qt-project.QtCore         	0x00000001079002b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
16  org.qt-project.QtGui          	0x000000010714fbdf QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3615
17  org.qt-project.QtGui          	0x000000010713513b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 203
18  libqcocoa.dylib               	0x000000010b35ff98 0x10b323000 + 249752
19  com.apple.CoreFoundation      	0x00007fff2040994c __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
20  com.apple.CoreFoundation      	0x00007fff204098b4 __CFRunLoopDoSource0 + 180
21  com.apple.CoreFoundation      	0x00007fff20409634 __CFRunLoopDoSources0 + 242
22  com.apple.CoreFoundation      	0x00007fff2040805c __CFRunLoopRun + 893
23  com.apple.CoreFoundation      	0x00007fff2040761c CFRunLoopRunSpecific + 563
[... 512 lines cut ...]
__LINKEDIT                       510.8M      107 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2496K        2 
__TEXT                           254.5M      377 
__UNICODE                          588K        1 
mapped file                      399.2M       31 
shared memory                     1268K       15 
===========                     =======  ======= 
TOTAL                              1.9G     1384 
TOTAL, minus reserved VM space     1.5G     1384 

Model: MacBookPro13,3, BootROM 429.140.8.0.0, 4 processors, Quad-Core Intel Core i7, 2.9 GHz, 16 GB, SMC 2.38f12
Graphics: kHW_IntelHDGraphics530Item, Intel HD Graphics 530, spdisplays_builtin
Graphics: kHW_AMDRadeonPro460Item, AMD Radeon Pro 460, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.8)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: Apple T1 Controller
USB Device: PlutoSDR (ADALM-PLUTO)
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.2
```