analogdevicesinc/scopy #1074: <Instrument>: <Mixed Signal View crashes when changing time base>

alex-t-nguyen, 2021-04-04:
OS Version: Windows 10 Version 1909
FW Version: 
SW Version: v1.2.0

Description of the bug:<Mixed signal view crashes when changing time base>
Steps to reproduce:
- Turn on Mixed signal view
- Change time base

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/Niem/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

adisuciu, 2021-04-05:
Can you try the latest version - https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc1 ?

alex-t-nguyen, 2021-04-05:
It appears to be working now when you change the time base. However, if I click autoset on the oscilloscope while mixed signal view is on it still crashes.

adisuciu, 2021-04-22:
Closing this as this issue is related to autoset - https://github.com/analogdevicesinc/scopy/issues/913

closed 2021-04-22 by adisuciu