analogdevicesinc/scopy #1076: <"ADALM2000>: <Version 1.3 Bug>

yazzy18, 2021-04-04:
**Environment:**
 - OS Big Sur: [e.g. iOS]
 - FW Version 
 - SW Version or git tag

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Power Supply'
2. Set Positive Power to 5V & Negative Power to -5V
3. Set Signal Generator CH-1 to Triangle, Vp-p=5V, Frequency = 50 Hz, Offset = 2.5V
4. Turn on Oscilliscope
5. Autoset CH-1 
6. notice that exported data is very skewed and not consistent 
6. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Have a consistent set of data values (i.e. the data is supposed to produce a x^3 graph)
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Additional context**

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/ali/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

Add any other context about the problem here.

adisuciu, 2021-04-05:
Does this only happen when you autoset ? Or when changing timebase / voltsdiv as well ?

yazzy18, 2021-04-05:
This happens when I auto set. I’ve also been noticing the oscilloscope has
been taking longer than usual to display the graph and measurements since
that crash. It also crashed twice on me yesterday.

adisuciu, 2021-04-06:
We're aware of some bugs in the autoset feature. We're working on fixing them, however, the fix requires some architecture changes which will not be done in time for v1.3.0. This bug appears on macOS and sporadically on Windows.

https://github.com/analogdevicesinc/scopy/issues/913

Can you screenshot the skewed nonconsistent data ? Could it be because the autoset fails and it sets some random v/div or timebase ? 

-Adrian

closed 2024-09-24 by adisuciu