analogdevicesinc/scopy #1084: General: Scopy crashes when it is reset in the Preferences

HRosete, 2021-04-07:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: 441fd19

**Describe the bug**
Scopy crashes when it is reset on the Preferences. Please see the video in the zip file below.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect M2K to Scopy.
2. Once connected, go to 'Preferences.'
3. Click 'Reset Scopy' on the lower-right corner of the UI.
4. Wait for the program to respond to see the error.

**Expected behavior**
Scopy must successfully reload and revert back to its default settings.

**Screenshots**
Please unzip this file to see the screen record.
[scopy1v3_reset_crash_gen_settings_edit_0.zip](https://github.com/analogdevicesinc/scopy/files/6269695/scopy1v3_reset_crash_gen_settings_edit_0.zip)

**Additional context**
[INI Files.zip](https://github.com/analogdevicesinc/scopy/files/6269740/INI.Files.zip)

DanielGuramulta, 2021-04-08:
I could't reproduce this bug on my Windows machine. Can you have a look at the following 2 scopy test builds and see if the issue is still showing up, thank you.
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38594250/job/g9m4ef8xj0x6yj6o/artifacts
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38594275/job/n7hi5w02xi1v4ia0/artifacts

Daniel

adisuciu, 2021-04-12:
I think this should fix it - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/dyee5dcpre1plp7l/artifacts - can you retest ?

adisuciu, 2021-04-21:
Does it crash with the latest ?

HRosete, 2021-04-29:
I believe this issue is related to issue #1089 in some ways. However, #1089 has now been confirmed fixed on c625d89. Sorry for the late response.

Also, tried to recreate this issue on c625d89 (Scopy v1.3.0-rc2) and is confirmed nonexistet/fixed. I'll be closing this issue now. Thank you.

closed 2021-04-29 by HRosete