analogdevicesinc/scopy #1086: Digital IO (UI): DIO indicator initially says "Individual" even if the IOs are grouped

HRosete, 2021-04-07:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: 441fd19

**Describe the bug**
The groups of DIOs 0-7 and DIOs 8-15 initially indicate "Individual" even if the DIOs are initially grouped. Please see screenshot below.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Digital IO'
2. Note the setting and orientation of each of the two DIO groups to see the error.

**Expected behavior**
The initially grouped DIOs must be labeled as "Group", not as "Individual."

**Screenshot**
[image]

DanielGuramulta, 2021-04-07:
Should be fixed with pull request: https://github.com/analogdevicesinc/scopy/pull/1081

jpineda3, 2021-04-28:
Verified fixed in v1.3.0-rc2.

HRosete, 2021-04-29:
> Verified fixed in v1.3.0-rc2.

Screenshot:
[image]

Closing this issue now.

closed 2021-04-29 by HRosete