analogdevicesinc/scopy #1088: General: White splash screen appears when opening scopy

tagoylo, 2021-04-08:
Environment:

OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
FW Version: v0.26
SW Version or git tag: 441fd19

Describe the bug
When opening Scopy there's a white splash screen that appears first before scopy finally loads. See picture and video below.

To Reproduce
Steps to reproduce the behavior:
Open scopy and wait until it starts loading.

Screenshots
[image]
[image]

DanielGuramulta, 2021-04-08:
I couldn't reproduce this bug on my Windows machine with the exact same scopy version and firmware version that you are testing. I took a look and found something that may be causing this. Can you run a test with this scopy: https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38593985/job/ayxx7s42c24haavy/artifacts and tell me if this bug is still showing up, thank you.

Daniel

tagoylo, 2021-04-09:
The white splash screen does not appear in this scopy version. But a quick black splash screen appears but only for less than a second, it's not noticeable.
[image]

HRosete, 2021-04-09:
Hello! To support Trecia's claim, here are some GIFs and screen record of Scopy:

(Scopy default theme)
[image]

(Scopy light theme)
[image]

https://user-images.githubusercontent.com/43806793/114184235-2d979980-9977-11eb-8ce8-b89d33d596ad.mp4

These are taken on the same environment -- Windows 10, 441fd19.

adisuciu, 2021-04-09:
Can you reproduce with the build Daniel sent ?  https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38593985/job/ayxx7s42c24haavy/artifacts

HRosete, 2021-04-10:
I'm sorry, my bad. I missed Daniel's comment. I can't fully reproduce this on the build he sent, especially on the 'default' theme, unlike we observed before. Here are the screen records of Scopy opening and switching between different themes:

**Scopy opening up using the 'default' theme**

https://user-images.githubusercontent.com/43806793/114255900-fc57b180-99e8-11eb-989d-25f113388143.mp4

**Scopy switching theme from 'default' to 'light'**

https://user-images.githubusercontent.com/43806793/114255911-12657200-99e9-11eb-9730-dca098342b68.mp4

**Scopy opening up using the previously set 'light' theme**

https://user-images.githubusercontent.com/43806793/114255932-30cb6d80-99e9-11eb-8e53-5ae635ca0c3d.mp4

**Scopy switching theme from 'light' to 'default'**

https://user-images.githubusercontent.com/43806793/114255949-4b054b80-99e9-11eb-98c6-8bdcf74c829f.mp4

closed 2021-05-05 by adisuciu