analogdevicesinc/scopy #1089: General: Scopy crashes when M2K is reconnected to the reopened application

HRosete, 2021-04-08:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: 441fd19

**Describe the bug**
Scopy crashes when M2K is reconnected to the reopened application. This doesn't happen if M2K is unplugged prior to reopening the program, and just replugging and connecting it to Scopy after it has opened. I tested this bug using the two scenarios stated below. To gain more information, please see the videos and INI files attached in the ZIP files.

**To Reproduce**
Steps to reproduce Scenario 1:
1. Open Scopy.
2. 'Connect' M2K to the program. Don't disconnect! Don't unplug M2K from your laptop.
3. Close the application.
4. Reopen Scopy.
5. Reconnect M2K to Scopy. Wait to see the error.

Steps to reproduce Scenario 2:
1. Open Scopy.
2. 'Connect' M2K to the program.
3. 'Disconnect.' Don't unplug M2K from your laptop.
4. Close the application.
5. Reopen Scopy.
6. Reconnect M2K to Scopy. Wait to see the error.

**Expected behavior**
Scopy must reopen successfully when M2K is reconnected to the application, whether M2K is reattached to the laptop/PC prior to opening the program or not.

**Additional contexts**
[INI Files Scopy Reopen Crash Scenario 1 - With Debug.zip](https://github.com/analogdevicesinc/scopy/files/6275908/INI.Files.Scopy.Reopen.Crash.Scenario.1.-.With.Debug.zip)
[INI Files Scopy Reopen Crash Scenario 2 - With Debug.zip](https://github.com/analogdevicesinc/scopy/files/6275911/INI.Files.Scopy.Reopen.Crash.Scenario.2.-.With.Debug.zip)

HRosete, 2021-04-08:
**Screen records**

https://user-images.githubusercontent.com/43806793/113969840-4ebd9480-9868-11eb-97ad-b2d630738fc5.mp4

https://user-images.githubusercontent.com/43806793/113970263-2edaa080-9869-11eb-90b9-0aceeb5e7285.mp4

adisuciu, 2021-04-12:
This version should fix it - https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/dyee5dcpre1plp7l/artifacts

adisuciu, 2021-04-12:
Can you retest ?

HRosete, 2021-04-12:
Sure, Adrian. Downloading it now.

HRosete, 2021-04-12:
I tested this issue using the scenarios above on the build you sent and Scopy reopens perfectly fine now. It didn't crash when I reopened it and I didn't replicate the bugs. I'd just like to note that M2K connected to Scopy more slowly than it used to on this build (a751ed5)

adisuciu, 2021-04-12:
First time after installation it always connects more slowly because it sets up the python environment for decoders. Does it open more slowly everytime you connect ?

HRosete, 2021-04-13:
No, just the first time after installing the Scopy build you sent. After I noticed it, I tried to connect and disconnect M2K multiple times and it connected faster than the first time. If it is what you said because it sets up the python environment for decoders, then I think it doesn't post as a problem. :) Thanks for clarifying Adrian!

HRosete, 2021-04-13:
Issue #1089 can't be replicated as of the build a751ed5.

HRosete, 2021-04-29:
This issue has been confirmed fixed on Scopy v1.3.0-rc2 (c625d89) on both of the scenarios stated above. Closing this issue now.

closed 2021-04-29 by HRosete