analogdevicesinc/scopy #1090: Logic Analyzer: Oscilloscope as External Trigger Source has conditions

tagoylo, 2021-04-08:
**Environment:**

- OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
- FW Version: v0.26
- SW Version or git tag: 441fd19

**Describe the bug**
When selecting oscilloscope as the external trigger source, sometimes the dropdown menu for conditions is enabled.

**To Reproduce**
Steps to reproduce the behavior:

1. Open Logic Analyzer
2. Click on the Trigger menu.
3. Turn ON external trigger.
4. Select Oscilloscope as Source. This time the conditions are disabled. 
5. Do not change the Source, leave it as oscilloscope. 
6. Turn OFF external trigger. 
7. Then turn it ON again. The source is still Oscilloscope but the Conditions are now enabled.

**Expected behavior**
Conditions should stay disabled when the oscilloscope is set as an external trigger source.

**Screenshots**
[image]

DanielGuramulta, 2021-04-08:
Should be fixed with pr: #1091

tagoylo, 2021-04-28:
Issue verified fixed in v1.3 RC2. Closing this item.

closed 2021-04-28 by tagoylo