analogdevicesinc/scopy #1094: <Instrument>: <Short description of the bug>

tadashi2014, 2021-04-10:
OS Version: macOS 10.15
FW Version: 
SW Version: e02cebe

Description of the bug:<description>
Steps to reproduce:
-
-

The ini files might be useful to reproduce the error.
The ini file is located at: /Users/tadashi/.config/analog.com/Scopy.ini
Please consider attaching it.

adisuciu, 2021-04-12:
Can you elaborate ?

tadashi2014, 2021-04-12:
With the adalm2000 connected, I selected a logic analyzer and tried to run it without connecting anything to the IO connector, and the application quit.
I have multiple USB drivers installed, which may be the effect.

adisuciu, 2021-04-12:
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/im3mrfswwg3oa1gb/artifacts 
Would you mind giving this version a try ?

tadashi2014, 2021-04-12:
The downloaded application seems to be working without problems.

adisuciu, 2021-04-12:
Awesome. This will be part of the new release coming soon. Stay tuned. I'm closing this for now

closed 2021-04-12 by adisuciu

tadashi2014, 2021-04-12:
OK. I'm looking forward to the release of the new version.