analogdevicesinc/scopy #1095: Voltmeter: Spikes in history plot, fluctuating DC and VRMS reading

jpineda3, 2021-04-11:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26 ](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: [441fd19](https://github.com/analogdevicesinc/scopy/commit/441fd19324d7256b64ede2551e4e7bff9e525a86)

**Describe the bug**
Fluctuating values and spikes in history plot even at DC mode. More frequent in Ch2. **Behavior is unpredictable.** There are times when the voltmeter would work just fine but fluctuations are present more often.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 2+ probe to positive supply (V+) and 2- probe to ground.
2. Go to Voltmeter and select AC mode for channel 2.
3.  Go to Signal Generator and set Ch1 to output a square wave with amplitude=5Vpp, frequency=800Hz, with phase=0; offset=0.
4. Enable Signal Generator and Voltmeter. 

**Expected behavior**
If by step 4 the spikes are not present, try to toggle the Signal Generator off-on.

**Screenshots**
https://user-images.githubusercontent.com/76926625/114300984-41690a00-9af5-11eb-8e55-d8852c882868.mp4

_Screenshots below are not from the scenario above. Included here to show similar behavior in ch1, and at DC mode.
[image]
_
[image]

**Additional context**
[voltmeter_fluctuations_ch2.zip](https://github.com/analogdevicesinc/scopy/files/6291544/voltmeter_fluctuations_ch2.zip)

adisuciu, 2021-04-13:
I cannot reproduce this ..

jpineda3, 2021-04-14:
Please try this instead.

1. Connect 2+ to W2 and 2- to ground.
2. Set Voltmeter Channel 2 in AC mode, gain Auto. **DO NOT RUN THE VOLTMETER YET.** 
3. Set Signal Generator Channel 2 to output a sinewave with amplitude 10Vpp, frequency 1kHz, no offset, no phase. **DO NOT RUN THE SIGNAL GENERATOR YET.** 
4. Go to the Voltmeter panel and run the voltmeter. The measurement in Channel 2 of the Voltmeter should be steady at 0VRMS.
5. Without leaving the Voltmeter panel (to properly observe the issues), run the Signal Generator from the list of instruments on the left by clicking the square next to it. **Spikes should start to appear in Voltmeter Channel 2.** 
6. **Without stopping the Voltmeter**, change the amplitude of the output of Channel 2 from 10Vpp to 3Vpp.
7. Go back to Voltmeter panel. VRMS readings should be steady at around 1VRMS.

I think this issue is related to the gain setting in the Voltmeter. Running Channel 2 in AC mode with no input signal would require High gain. Running the Signal Generator with 10Vpp output would require the voltmeter to transition to Low gain mode. You could observe in Step 5 that the gain mode in Channel 2 is flipping between High and Low.

This is not an issue for Channel 1 when Channel 1 is running alone. But when both channels are enabled and either channel is exposed to changing signal levels such that the spikes are present, both channels are affected. When both channels are fed with the signals requiring the same gain level, however, the voltmeter would work fine.

Please see this recording below.
https://user-images.githubusercontent.com/76926625/114674614-fd1a7b80-9d39-11eb-9f8b-e0e5fdbb4ca1.mp4

adisuciu, 2021-04-21:
Thanks for the analysis - it was useful in figuring out the problem. I think I found the issue, can you give this another try please  ?
https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38810091/job/oo4yjqwlo30bn80m/artifacts

-Adrian

jpineda3, 2021-04-22:
The build looks good. I did a quick test on these combinations. No fluctuations or spikes were observed. 
[image]

adisuciu, 2021-04-22:
Awesome, thanks !

closed 2021-04-22 by jpineda3