analogdevicesinc/scopy #1096: Pattern Generator: Selecting UART pattern crashes Scopy

tagoylo, 2021-04-12:
**Environment:**

- OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
- FW Version: v0.26
- SW Version or git tag: 441fd19

**Describe the bug**
When selecting UART as the pattern for a channel. Scopy then crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable a channel
2. Double Click on the channel handle
3. Select UART as pattern. Scopy crashes.

adisuciu, 2021-04-12:
Cannot reproduce - can you also try it with this version ? https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/dyee5dcpre1plp7l/artifacts

adisuciu, 2021-04-21:
Does it crash with the latest ?

tagoylo, 2021-04-21:
> Does it crash with the latest ?

No, it now doesn't crash with the latest. >https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38787957 used this build. Also checked with the new firmware and it doesn't crash.

adisuciu, 2021-04-21:
alright, thanks !

tagoylo, 2021-04-28:
Issue verified fixed in v1.3 RC2. Closing this item.

closed 2021-04-28 by tagoylo