analogdevicesinc/scopy #1099: Logic Analyzer: UART decoder not on the list

tagoylo, 2021-04-12:
**Environment:**
- OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
- FW Version: v0.26
- SW Version or git tag: 441fd19

**Describe the bug**
UART is missing from the list of decoders.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Logic analyzer
2. Click on select a decoder to add
3. Look for UART,  it can be noticed that UART is not on the list.

**Expected behavior**
UART should be on the list.

**Screenshots**
[image]

adisuciu, 2021-04-12:
Cannot reproduce - can you also try it with this version ? https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/dyee5dcpre1plp7l/artifacts

tagoylo, 2021-04-12:
> Cannot reproduce - can you also try it with this version ? https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38632761/job/dyee5dcpre1plp7l/artifacts

Tried this version and UART is still not on the list. Also tried uninstalling scopy first before installing this version but still the same issue occurs.

adisuciu, 2021-04-12:
Can you browse to your Scopy installation directory/decoders and check if you have an uart folder in there and inside that folder tehre should be pd.py and __init__.py ?

-Adrian

tagoylo, 2021-04-12:
Yes, uart folder is present as well as the pd.py and init.py files

adisuciu, 2021-04-20:
Can you try this build ? https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38787957

tagoylo, 2021-04-21:
> Can you try this build ? https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/38787957

The uart is now showing on this build.

tagoylo, 2021-04-28:
Issue verified fixed in v1.2 RC2. Closing this item.

closed 2021-04-28 by tagoylo