analogdevicesinc/scopy #1101: Logic Analyzer: Saved export file has redundancy in file extension (.csv.csv)

tagoylo, 2021-04-12:
**Environment:**

-  OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
- FW Version: v0.26
- SW Version or git tag: 441fd19

**Describe the bug**
When exporting and checking the saved file. The saved file has the following filename, **_filename.csv.csv_**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to logic analyzer
2. Click on export button
3. Add a filename, ex. sample
4. Check the file that is saved.
It can be observed that together with the filename a .csv is added aside from the file extension.

**Screenshots**
[image]
[image]

adisuciu, 2021-04-12:
Fixed here - https://github.com/analogdevicesinc/scopy/pull/1105

tagoylo, 2021-04-28:
Issue verified fixed in v1.3 RC2. Closing this item.

closed 2021-04-28 by tagoylo