analogdevicesinc/scopy #1102: Oscilloscope: Minor bug on parsing exported CSV files, a semicolon is included

HRosete, 2021-04-12:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: 441fd19

**Describe the bug**
A semicolon (;) is included at the start of the first 7 lines of the exported CSV file from the instrument. Please see the screenshot below.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Scopy and load the profile inside this ZIP folder: [sample_scopy_profile.zip](https://github.com/analogdevicesinc/scopy/files/6295198/sample_scopy_profile.zip)
2. Run 'Signal Generator' and 'Oscilloscope.'
3. Go to the 'General Settings', turn on 'Export All', and finally click 'Export.'
4. Save the file as 'all_channels.csv.'
5. Open the CSV file and take note of the first seven (7) rows to see the minor error.

**Expected behavior**
Although this will not affect the data exported, there must be no semicolons before the texts.

**Screenshots**
[image]

adisuciu, 2021-04-12:
That is the correct format right now .  https://github.com/analogdevicesinc/scopy/blob/master/src/filemanager.cpp#L103

HRosete, 2021-04-13:
Oh. Copy. I'll close this issue now then. Thank you for the clarification.

closed 2021-04-13 by HRosete