analogdevicesinc/scopy #1106: Signal Generator: Persistent CH1 indicator (orange line) even when channel is turned off.

kimpaller, 2021-04-13:
**Environment:**
 - OS: Windows 10
 - FW Version: https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26
 - SW Version or git tag: https://github.com/analogdevicesinc/scopy/commit/441fd19324d7256b64ede2551e4e7bff9e525a86

**Describe the bug**
CH1 indicator (orange line) does not disappear even when CH2 is turned off. It will just go back to a straight line at 0.00V after CH1 is switched off.

Before CH1 is turned off:
[image]

After CH1 is turned off:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Signal Generator'.
2. Click on CH1 (orange circle) to activate the channel.
3. Click again on CH1 to deactivate channel.
4. Observe orange straight line at 0.00V.

**Expected behavior**
After deactivating the channel, the orange line (CH1 indicator) shuld disappear just like CH2 indicator (blue line).

adisuciu, 2021-04-13:
Fixed https://github.com/analogdevicesinc/scopy/pull/1107

kimpaller, 2021-04-29:
Confirmed fix on:

OS: Windows 10
FW Version: https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.27
SW Version or git tag: [v1.3.0-rc2](https://github.com/analogdevicesinc/scopy/releases/tag/v1.3.0-rc2)

closed 2021-04-29 by kimpaller