analogdevicesinc/scopy #1111: General: Automatic M2K FW update doesn't work

HRosete, 2021-04-19:
**Environment:**
 - OS: Windows 10 Enterprise 64-bit (10.0, Build 17763)
 - FW Version: [v0.26](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.26)
 - SW Version or git tag: a751ed5

**Describe the bug**
The automatic M2K FW update doesn't work. On April 14th (5 days upon posting this), M2K's new firmware version [v0.27](https://github.com/analogdevicesinc/m2k-fw/releases/tag/v0.27) has been released. My instrument still has the v0.26 FW. When I enabled the automatic update checking on the Preferences and opened Scopy, it says that my "Firmware is up to date" even if it isn't. Please see the pictures below:

**Screenshots**
[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. There are two ways to enable the automatic update feature.
1.1. Click 'Yes' on the dialog box prompt upon opening Scopy. It will automatically enable the corresponding checkbox on the Preferences.
[image]

1.2. If you've initially clicked 'No' on the prompt, enable it by clicking the "Enable automatic update checking" checkbox on the Preferences.
[image]

2. Attach M2K loaded with a v0.26 firmware.
3. Click on 'Home.'
4. Click on the M2K device icon.
5. Check on the message below the 'Connect' button and see the error.

**Expected behavior**
Scopy must correctly compare whether the FW loaded on the M2K device is the same as the newly released FW version on GitHub. Also, it must accurately state whether the FW on the device is truly up to date or not.

adisuciu, 2021-04-20:
It should work now. I updated the file on the server.

HRosete, 2021-04-29:
This issue is now confirmed fixed on Scopy v1.3.0-rc2 (c625d89). Please see screenshot for proof:

[image]

Closing this issue now.

closed 2021-04-29 by HRosete