analogdevicesinc/scopy #1119: Logic Analyzer: Delay doesn't seem to work

tagoylo, 2021-04-28:
**Environment:**
 - OS: Windows 10
 - FW Version : v0.27
 - SW Version or git tag:  v1.3.0-rc2  c625d89

**Describe the bug**
Adding delay to the plot does not seem to add delay to the plot. Also, is delay supposed to be disabled when in stream mode?

**To Reproduce**
Steps to reproduce the behavior:
1. In pattern generator, enable a channel and set it to any pattern to generate a signal
2. In logic analyzer monitor that channel 
3. Add delay to the plot.
4. Observe that there are no changes to the plot.

[image]

5. Also when changing to stream, delay becomes disabled. is this expected? 
[image]

adisuciu, 2021-04-28:
Yes, it's supposed to be disabled in stream mode
You did not set a trigger. Delay sets the trigger delay - basically the trigger will occur on the blue vertical line.

tagoylo, 2021-04-28:
> Yes, it's supposed to be disabled in stream mode
> You did not set a trigger. Delay sets the trigger delay - basically the trigger will occur on the blue vertical line.

Oh right. Thanks for the clarification. Closing this item.

closed 2021-04-28 by tagoylo