analogdevicesinc/scopy #1120: Oscilloscope: Plot labels

CJW-1014, 2021-05-01:
**Environment:**
 - WIn10: [e.g. iOS]
 - v1.2.0 - v1.2.0

**Describe the bug**
The preferences for the oscilloscope has an "Enable labels on the plot" check box.  With it checked, I see the voltage of the grid for the last channel enabled but not both.  There is room for both to be shown.  The grid moves to the right when the second channel labels are shown.  How do I get the labels for both channels to be shown?  SInce they are of different colors there should be no confusion.

Both channels should have associated labels.  It is confusing trying to discover how to make the labels for a specific channel to display.

[image]

[image]

gastmaier, 2023-03-15:
Qwt seems to resize the plot according to the number of axis enabled (with QwtPlot::setAxisVisible method) ([demo](https://gist.github.com/gastmaier/bb20dac732ba48765195cf7a766f569d) ([cmake](https://gist.github.com/gastmaier/894c8875a9958fb293a2fcd9a9b54554) helper)).
However, when done with scopy:
[1] Labels get superposed.
And also:
[2] We only keep track of the current enabled with DisplayPlot::d_activeVertAxis.

Considering:
[1]  labels might be set to 0-width somewhere else or the plot might be "squishing" them if set to something like min-width 100%, we would have to check where the issue is (does [GammaRay](https://github.com/KDAB/GammaRay), or some other inspector work with scopy?).
[2] we could keep track of more axis in an array and limit to the last three to be visible in the plot; this tracking could happen in the same[ DisplayPlot::setActiveVertAxis](https://github.com/analogdevicesinc/scopy/blob/f2245fdc2ab0eac2a83d29ef508f86e1d9a02a34/src/DisplayPlot.cc#L1788) method as d_activeVertAxis.

adisuciu, 2023-03-15:
Yes, Gammaray kinda works with scopy, I managed to attach to the application once it started. 
I'll investigate the rest of your points and get back at you ..