analogdevicesinc/scopy #1135: Oscilloscope: Missing rise time, period, frequency, duty cycle measurements

sja1440, 2021-06-18:
**Environment:**
 - OS: Ubuntu 20.04 (fully updated)
 - FW 0.27 
 - SW 1.3.0 dda5ec6 (Appimage)
 - ADALM2000 connected to Scopy via USB.

**Describe the bug**
The measurements related to rise time, period, frequency, duty cycle etc never seem to be  populated even though they are enabled.  Please see the attached screenshot.
[image]

I have tried resetting scopy but nothing changes.

Given that I am a new scopy user, I suppose that it is possible that there is some config parameter that needs to be set - if there is then I have searched but not found it.

If you need me collect specific evidence please do not hesitate to ask.

adisuciu, 2021-06-18:
Yes, because it is impossible to calculate based on your waveform. There is no periodicity in the waveform on the screen, and it is impossible to compute dutycycle(in percent of the whole waveform) because it is not all being acquired. 
Up the timebase and retry - make more of the waveform visible on screen.

sja1440, 2021-06-18:
Thanks, that makes sense. 

By increasing the value of the timebase  I see that the rise time measurements etc are indeed calculated for channel 1 (orange) which is across the scopy Signal Generator (square wave amplitude 3.3V at 37kHz).

 However, the measurements do not appear for channel 2 (purple) which is measuring the potential difference across a resistor on the load side of a MOSFET LED driver. Vpp of channel 2 is around 500mV. I have tried timebases all the way up to showing 30 pulses at a time: channel 1 continues to show rise time etc whilst channel 2 never does.

By eye, the periodic nature of channel 2 seems to track pretty well channel 1 and so I find it hard to understand why rise time etc are not being calculated also for channel 2.  I guess that the algorithm used to identify periodicity does not recognise my channel 2 waveform as a periodic signal.  Here is screen shot showing channel 1 vs channel 2:
[image].  This does seem strange to me.

adisuciu, 2021-06-18:
If you view more of the waveform, you can see it - it works for me - 
[image]

Can you confirm that it doesn't work this way for you ?

sja1440, 2021-06-18:
I am sorry but how do you get the yellow and purple measurements being interleaved like that?

By the way, it seems to work when I Single shot the oscilloscope for channel 2 even with as few as pulses on the screen.  Though if I increase the time base (more pulse on screen), then occasionally the rise time is not shown.

adisuciu, 2021-06-18:
Select channel 1 in bottom left, select measurements menu, activate individual measurements (for channel 1), select channel 2, activate individual measurements for channel 2, select channel 1 again, activate some more measurements :) 

By the way, it seems to work when I Single shot the oscilloscope for channel 2 even with as few as pulses on the screen. Though if I increase the time base (more pulse on screen), then occasionally the rise time is not shown. - this does not reproduce for me . And it is not clear for me what you are looking at since in the screenshot you are zoomed in.

sja1440, 2021-06-18:
OK I now see how you got those measurements interleaved.  Here is a screenshot of single "shot" where channel 1 rise time etc is shown and channel 2 is not:
[image]

adisuciu, 2021-06-18:
Yes, I see it now .. It's a bug because the waveform never crosses zero volts .. We'll invesgitage further

sja1440, 2021-06-18:
Aha. Up to now I have been triggering on channel 1 (3.3 Vpp).  If I trigger instead on channel 2 (Vpp approx 500mV) then the rise time is nearly always calculated also for channel 2!  Here
[image]
 is a screen shot: