analogdevicesinc/scopy #1136: Consider using -mmacosx-version-min=10.13 for the 10.14 builds - seems to run fine.

albhm, 2021-06-19:
For those still using older os x versions: The 10.14 builds of scopy show a 'prohibited' traffic sign over the icon and can't be started by clicking. They (seemingly) run fine when opening the binary within the .app package. This stackoverflow [post](https://stackoverflow.com/questions/39310079/mac-app-package-backwards-compatibility-error-you-cant-use-this-version-of-t) shows how to set the min mac os version for the binary.

Another quick way to have a nice double-click-to-launch version for os x 10.13 (0x0a 0x0d) is to open the scopy binary in a hex editor and search for the first occurrence  of 0e0a00000e0a and change it to 0d0a00000d0a - voila traffic sign gone.

AlexandraTrifan, 2021-06-24:
Hi,

Could you test this nightly build ( https://ci.appveyor.com/project/analogdevicesinc/scopy/builds/39731230/job/2qm4m1cxi2qee46o/artifacts )? 
We tested this on 10.14 and set the min macos version to 10.13.
If you could confirm this works fine, we'll merge it into master.

Thanks!
-Alexandra

albhm, 2021-06-25:
Hi Alexandra,

I can confirm that the above build works out of the box on my 10.13 system. Every instrument runs and I could not notice anything unusual. The only thing that does not work is the 'demo' mode - it shows 'calibration failed' - but at least I can happily live with the 'real hw mode' only. 

Thanks!

AlexandraTrifan, 2021-06-25:
Hi,

The calibration in demo mode is a known issue. The demo mode is by default calibrated. In order to be able to use all the instruments in demo mode, you can work around this problem in the following way:
- disconnect from the Demo device;
- go to Preferences and check the "Skip calibration if already calibrated" option;
- connect to the Demo device; (the status after connect will be: "calibration skipped because already calibrated";

Thank you!
-Alexandra

closed 2021-07-05 by adisuciu