analogdevicesinc/scopy #1139: Quetion on Signal Gen Documentation on Wiki

damercer, 2021-07-01:
Hi:
I have a question / comment on the documentation for the Signal Generator instrument on the Scopy ADI Wiki  page. Don't know if this is the right place to post this vs on EZ.

The comment is:
There does not seem to be any mention of the additive noise feature and how it works. Or am I missing something?

The question is:
Is the added noise waveform calculated once and simply added to the waveform buffer data or is a new random value calculated on the fly sample by sample and added as the waveform buffer is played out?
The first way is least computationally intensive and done on the Host and the second is more computationally intensive and done presumably in the FPGA.

So How is it done and shouldn't that be documented in the Sig Gen Wiki page?

Thanks

Doug

adisuciu, 2021-07-02:
This is the way it's currently implemented: "Is the added noise waveform calculated once and simply added to the waveform buffer data"

I'll mention this in the wiki ..

Next step we want to do is actually stream the data which would improve noise randomness (because it is computed on the fly by the computer) but it would only work for low sample rates.

damercer, 2021-07-02:
Great, That's what I figured was happening. That is often sufficient when looking in the time domain but not so much when looking in the frequency domain. The explanation will help users better understand the spectrum of the noise like the one in #1130.

Doug

adisuciu, 2021-07-05:
I updated the doc. I'll close this if that's ok.

closed 2021-07-05 by adisuciu