analogdevicesinc/scopy #1142: ADALM2000 not getting detected by Scopy while using a Dell docking station

vsethia-adi, 2021-07-16:
I have connected ADALM2000 through a USB cable to the Dell docking station.
READY LED glows permanently While the LED1 blinks frequently.
However, the ADALM2000 is not detected by the Scopy although the Device Manager is able to detect it as M2k serial device.
Any permanent solution is very much appreciated, as nowadays the laptop is going to have two or lesser usb ports and we will have to use USB hub in field or  at workshop as all of the existing ports will be in use one way or the other.

adisuciu, 2021-07-21:
We've been seeing the same behavior unfortunately with no permanent solution right now. Just to confirm, what docking station and laptop are you using ?

-Adrian

vsethia-adi, 2021-07-21:
It is Dell Precision 5530 and Docking station TB16.

mhennerich, 2021-07-21:
Does it make a difference if you plug into the USB SS (blue) or HS ports on the back?
Have you tried using a different cable?

vsethia-adi, 2021-07-21:
No difference irrespective of the SS or HS USB ports on the docking station.
I did notice that sometimes the Scopy is able to detect the kit momentarily and lost the connection unless the Scopy is relaunched again.

teoperisanu, 2021-07-22:
Did you install the updates provides by Dell. There are some important updates such as:
- Dell Precision 5530 System BIOS
- Intel Thunderbolt Controller Driver
- Dell Thunderbolt Dock TB16 Firmware Update Utility
and others

The updates that can be found on Dell Support website and some of them can be installed via the Dell Update App.

vsethia-adi, 2021-07-23:
Yes, the above DELL drivers are installed.

I did an install of Scopy of 1.2 verson (downgrade form v1.3) and it was able to detect and connect the hardware. I was able to use them for short time. but the connection was patchy and the Scopy would crash several times. At times, I have to reset the Scopy software in the preferences, as it fails to recognize the hardware. I am using the following options - Oscilloscope, and Logic Analyzer.

teoperisanu, 2021-07-26:
Just out of curiosity, have you tried to connect via IP (192.168.2.1)?

vsethia-adi, 2021-07-26:
let me know how to use that.

teoperisanu, 2021-07-26:
Click on the '+' button, then enter the IP (192.168.2.1) in the URI field, then 'Connect', 'Add' and again 'Connect'

vsethia-adi, 2021-07-27:
No, it does not connect using the IP address.

rgetz, 2021-07-27:
what shows up in device manager? it should look like:
https://wiki.analog.com/_media/university/tools/pluto/drivers/device_manager_m2k_installed.png

vsethia-adi, 2021-07-27:
[image]
Snapshot attached. 
I see that sometimes it shows up as a normal USB device.
[image]

rgetz, 2021-07-27:
Does it help to plug the USB power supply on the M2k into a USB power supply (while you are plugged into the dock?)

vsethia-adi, 2021-07-27:
I did not understand your last question.

adisuciu, 2021-07-29:
Robin is asking if you see similar behavior when connecting an additional USB cable to an external power brick (not to the dell docking station) to the rightmost USB connector. This additional USB connector is used for power delivery only.

vsethia-adi, 2022-02-04:
The workaround in this case is 
1. Go to Device Manager tool in Windows OS
2. Go to Universal Serial Bus Controllers
3. Select ASMedia USB Root hub, Right click on its properties
4. uncheck the "Allow the computer to turn off this device to save power" in the Power management tab
5. Reboot.

It works for me.. Do confirm if it fixes the issues for you.  I have the Dell laptop model 5530 and docking station TB16.
ADALM2000 is connected via TB16. No issues if connected directly to laptop.

adisuciu, 2022-02-04:
Awesome - we updated our troubleshooting page - https://wiki.analog.com/university/tools/m2k/scopy-troubleshooting?rev=1643979322#make_sure_the_system_doesn_t_limit_the_usb_power_consumption

closed 2022-02-04 by vsethia-adi