analogdevicesinc/scopy #1143: Oscilloscope: vertical position modifies when the window is resized

AlexandraTrifan, 2021-07-30:
The Oscilloscope vertical position value changes if some operations are done (examples: resizing the window; enabling the Measurements)

First posted here: https://ez.analog.com/adieducation/university-program/f/q-a/547707/scopy-v-1-30-fw-v0-27-adalm2000-o-scope-position-setting-question

damercer, 2022-03-08:
Hi, just checking in to see if there is an unreleased version (on AppVeyor?) that fixes this **annoying** issue? I'm using the official Jun 14 2021released version (1.3.0?) which came out just before this issue was posted. Also when will a new released version with this fixed be available. I hope this issue hasn't fallen through the cracks.
Doug

adisuciu, 2022-03-10:
Will be fixed here - https://github.com/analogdevicesinc/scopy/pull/1211

damercer, 2022-03-15:
Where can I down load a copy of the Windows installer with this fix to check it out? Looks like various jobs on AppVeyor built from master (after this was added) have run but I don't see any artifacts i.e. a Windows installer?
Thanks Doug

AlexandraTrifan, 2022-03-15:
Hi,

You can get the installer for Windows here: https://github.com/analogdevicesinc/scopy/actions/runs/1979778614 . 
I think that you need to be logged into Github to be able to download installers from the "Artifacts" section. If that's an issue, please let me know and I'll send it to you.

-Alexandra

tagoylo, 2022-05-04:
Unable to reproduce this in v1.4.0 rc2. Tried to enable measurements and resized the window but the vertical position value stays non-zero. Closing this item.

closed 2022-05-04 by tagoylo