analogdevicesinc/scopy #1148: PatternGenerator: investigate SPI behavior

adisuciu, 2021-08-23:
There are some suggestions that the patterngenerator SPI functionality is not working correctly.

https://ez.analog.com/adieducation/university-program/f/q-a/548567/incorrect-spi-framework-in-pattern-generator-in-scopy

This description should help with implementing the fix if needed. 

[image]

Yamakaja, 2023-03-13:
Just as a datapoint, this output also doesn't look correct, the last falling clock edge is missing
[image]

lysynski, 2023-08-24:
I'm surprise this is still an open issue, the screenshot show that the behaviour is incorrect.
Here's another showing incorrect behaviour.
[image]
Hopefully this gets address soon.

AlexandraTrifan, 2024-12-11:
Hi!

The initial release candidate for the new Scopy v2.0.0 is now available in the official releases page: https://github.com/analogdevicesinc/scopy/releases/tag/v2.0.0-beta-rc1. This candidate includes a lot of new features in Scopy while bringing multiple bug fixes for the ADALM2000 plugin and underlying libm2k library.

The pull request regarding this issue is: https://github.com/analogdevicesinc/scopy/pull/1496
Please reopen the issue if the bug still appears?

Thank you!