analogdevicesinc/scopy #1154: Possible enhancement for AWG signal genertor controls

damercer, 2021-09-03:
Here is a suggestion for a possible enhancement for the Scopy Signal (AWG) Generator amplitude controls. Since the AWG outputs include a 50 ohm series resistor the actual output voltage seen at the load connected to he AWG will be different than the set value based on the value of the load vs the 50 ohms.

Some function/arbitrary waveform generators (like Keysight’s) let you enter a value for the expected (connected) load resistance, RL, in the range of 1 ohm to 10 k ohms, or infinite. The default value is 50 ohms (or could be infinite) to match the internal value. If you change the value for this parameter, the function generator automatically adjusts the internal voltage it produces to take into consideration the voltage divider formed by the internal 50-ohm resistor and your load resistance such that VOUT is equal to your voltage setting. This adjustment applies to both the DC offset and AC portion of the function generator output signal.

In the case of M2k there would be a upper limit on how much "correction" is possible given the internal power supply voltage for the AWG driver Amps.

Just something that might make Scopy work more like some bench instruments (if that if the look an' feel we are going for).

Doug

adisuciu, 2022-03-10:
Implemented in https://github.com/analogdevicesinc/scopy/pull/1210

jpineda3, 2022-05-04:
This is verified working in v1.4.0 rc2 https://github.com/analogdevicesinc/scopy/commit/cca49b2ba901f8d8c40327c689761c87db368fa9

AC and DC output signal are appropriately scaled. The amplitude of the output signal is doubled at the minimum load resistance of 50ohm as expected.

closed 2022-05-04 by tagoylo