analogdevicesinc/scopy #1157: Error Check: Could I get the checksum of the file 'debug-v1.3.0-Windows-64bit.zip'

rahulanalog, 2021-09-14:
I have been tried to install the scopy software and due to some issue I am getting a weird downloads happening. Could you send me the checksum if the file so that I could verify the file ?

Please help how I could verify the file before opening it.

adisuciu, 2021-09-14:
If you want to install the windows version of Scopy, this is the download you want https://github.com/analogdevicesinc/scopy/releases/download/v1.3.0/scopy-v1.3.0-Windows-setup.exe

The file you mentioned only contains debug symbols that are useful if the application crashes and further investigation is required

hellosemiconductor, 2021-09-15:
Yes, Sorry I needed the checksum of the file that you mentioned.
[https://github.com/analogdevicesinc/scopy/releases/download/v1.3.0/scopy-v1.3.0-Windows-setup.exe](https://github.com/analogdevicesinc/scopy/releases/download/v1.3.0/scopy-v1.3.0-Windows-setup.exe)

adisuciu, 2021-09-15:
> adisuciu@DESKTOP-UT41HHH:/mnt/c/Users/Adi/Downloads/tmp$ sha256sum scopy-v1.3.0-Windows-setup.exe
> 390455c790f333d7019b247452b13495d7514d3600319035eea10d796aa9f3af  scopy-v1.3.0-Windows-setup.exe

hellosemiconductor, 2021-09-15:
Thanks a lot

closed 2021-09-15 by adisuciu