analogdevicesinc/scopy #1169: Spectrum Analyzer: Can not zoom in on y-axis

davidparent, 2021-11-04:
Windows 10 FW=.28 SW=V1.3.0 dda5ec6

One can not zoom in on the y-axis more than a range of 10 on VRMs or Vpeak.  I sued to be able to zoom into a range of 1. (V1.2)

Set a 1k sin wav with a 1V amplitude and run.  Got to specvtrm analsyer and change sweep uintil ne can see the 1kHz peak Change y-axis to Vpeak or Vrms. Adjust top and bottom and range to zoom in.    
One can only zoom in to a min of 10.  Can not set tp and bottom closer than 10.

**Expected behavior**
Use to be able to zoom into a range of 1

 
[image]

ioanachelaru, 2021-11-09:
Can you download [scopy-x86_64-setup.exe](https://github.com/analogdevicesinc/scopy/actions/runs/1439644639)  and let me know how it works?
Thanks in advance

-Ioana

davidparent, 2021-11-09:
Hi Ioana,
It works by setting the scale factor to what you want first.  Thanks!
BTW when I start the program I get this error:
[image: image.png]
Best,
-Dave

ioanachelaru, 2021-11-10:
The image isn't available, but I believe you are referring to a problem with libsigrokdecode. We are aware of it and it will be solved soon. 

-Ioana

davidparent, 2021-11-10:
Yep that was it. Thanks so much.

closed 2021-11-17 by ioanachelaru