analogdevicesinc/scopy #1181: ADALM2K: Scopy/Oscilloscope - Add More Attenuation Options

dstover-adi, 2021-11-20:
Scopy probe attenuation options are limited to 0.1/1/10/100x. This selection leaves some users in the cold--Our diff probes for example offer only 50/500x attenuation.

Please add more user control over probe attenuation.

adisuciu, 2021-11-23:
What are some common options ? Or you think this should be editable ?
Most of the scopes I've used you either select out of a very limited range (like 3 values) or they autodetect the attenuation using external circuitry.

dstover-adi, 2021-11-23:
Hi Adrian,

The scopes I’ve used allow attenuation in ratio or dB. In ratio, common options include 1, 2, 5 in decades: 0.01-0.05, 0.1-0.5, 1-5, 10-50, 100-500. These are also the options we’re used to on the old Agilent MSO scopes we use.

An editable field would offer the most flexibility if it isn’t too much of a bother.

Best,
David

[External]

damercer, 2021-11-25:
I don't know if this has made it to you all yet or not but Robin @rgetz  and I have discussed the whole external attenuation feature in Scopy. What is needed is the ability for the user to enter an arbitrary value for the attenuation (gain) and also potentially offset. Rather than a long (expanding) list of fixed values to choose from. A: None of these fixed values will ever be exactly correct for a given external "probe". B : The 10X passive probe we are about to offer with the M2k BNC adapters is only exactly 10 when the input resistance is exactly 1 Meg. The M2k input divider is not, by the design values in the schematic, exactly 1 Meg. The actual gain using the 10 X probes is a few % off and we need a way in Scopy for the user to adjust (calibrate) this error out, if they care to.

I recommend you rethink the external attenuation user interface to simplify it by just having the user enter a number for gain and offset. If the probe is 10X then they enter 10. If they what to adjust for any small error they enter 10.1 or whatever.

Just my (experienced) 2 cents.

Doug

adisuciu, 2022-03-10:
Implemented in https://github.com/analogdevicesinc/scopy/pull/1185

jpineda3, 2022-05-04:
This is verified working in v1.4.0 rc2 https://github.com/analogdevicesinc/scopy/commit/cca49b2ba901f8d8c40327c689761c87db368fa9
 
The volts/div adjusts accordingly when the probe attenuation is updated, with no limit to the allowed attenuation values.