analogdevicesinc/scopy #1188: compile problem

jsmith173, 2021-12-01:
**Environment:**
 - OS: Ubuntu 20.04

**Describe the bug**
1. scopy/CI/appveyor/install_qt_ubuntu_20.sh
There is a missing space in libqt5svg5-devqttools5-dev after libqt5svg5-dev

2. scopy/CI/appveyor/install_ubuntu_20_deps.sh failed at build_qwt

QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
Building qwt - branch qwt-6.1-multiaxes
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/attila/qwt/src/src.pro ) && make -f Makefile
make[1]: Entering directory '/home/attila/qwt/src'
compiling qwt_symbol.cpp
qwt_symbol.cpp:20:10: fatal error: qsvgrenderer.h: No such file or directory
   20 | #include <qsvgrenderer.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.

libqt5svg5-dev is installed

AlexandraTrifan, 2021-12-02:
Hi,

I will fix the build script to include the missing space. 
After modifying the script and adding the space, can you confirm that both these packages were correctly installed: libqt5svg5-dev qttools5-dev ?

Could you try to also install libqt5opengl5 and test again?

Thanks!
-Alexandra

jsmith173, 2021-12-03:
1. The packages correctly installed. qwt is compiled now. You should mention in the documentation
export QT_SELECT=qt5 anyway qwt won't compile
2. libm2k: ENABLE_EXAMPLES no longer exists you should use BUILD_EXAMPLES
3. libsigrokdecode does not compile
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
configure.ac:33: warning: macro 'AM_PROG_LIBTOOL' not found in library
autoreconf: configure.ac: tracing
autoreconf: configure.ac: creating directory autostuff
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:49: error: possibly undefined macro: AC_LIBTOOL_WIN32_DLL
  	If this token and others are legitimate, please use m4_pattern_allow.
  	See the Autoconf documentation.
configure.ac:51: error: possibly undefined macro: AM_PROG_LIBTOOL
autoreconf: /usr/bin/autoconf failed with exit status: 1

AlexandraTrifan, 2021-12-03:
For the libsigrok issue please try to install "libtool" and check if that helps.
So qwt is compiled now, after installing the libqt5opengl5 package?

Thank you!
-Alexandra

jsmith173, 2021-12-03:
Yes qwt compiled when I add this to .profile
export QT_SELECT=qt5

AlexandraTrifan, 2021-12-07:
Hi, 

Any updates on libsigrokdecode?

-Alexandra

jsmith173, 2021-12-08:
Installing libtool helped

closed 2022-01-11 by adisuciu