analogdevicesinc/scopy #1222: Version 1.4.0 testing

damercer, 2022-03-19:
**Environment:**
 - OS: Win 10
 - FW latest 
 - SW 1.4.0
Home screen:

[image]

While testing new version 1.4.0 candidate I found this issue in the Scope vertical range controls: With the AWG W1 generating a +5 to - 5V sine wave connected to Scope input 1+ and the vertical range set to 500 mV/Div with the scope running (screen updating) I see the sine wave going over range. (beyond the +2.5 to -2.5 grid). When I click on the + button to increase range to 1 V/Div the grid changes but the sine wave is now clipped at +3/-3V as if the hardware gain (input divider) has not been changed (remember Scope is running). As shown below:

[image]

If I again click the + range button to 2 V/Div the hardware gain is changed and I see the full +5 to -5 swing of the sine wave (no clipping).

[image]

Clicking the - button to go back to the 1 V/Div range and the clipping is now gone on this range.

[image]

**Expected behavior**
The HW gain should be changed on the fly while Scope is running as the 500 mV/Div range is increased to 1V/div.

adisuciu, 2022-03-22:
I think this should fix it - https://github.com/analogdevicesinc/scopy/pull/1227

tagoylo, 2022-04-29:
Issue not reproducible anymore in v1.4.0 rc2. Closing this item.

closed 2022-04-29 by tagoylo