analogdevicesinc/scopy #1224: ADALM2000: scopy android system requirements not met ?

jpnielsen, 2022-03-20:
I'd like to run Scopy on my tablet "PRODVX APPC-15X" - an rk3288 chipset 15" tablet. 1920x1080 resolution screen.
It's running Android 8.1.0 - and unfortunately cannot be upgraded to a newer android version.
The filesystem on the ADALM2000 device is showing up in android.
I downloaded the apk, and tried installed it, but I only get "Installing" .. and then "App not installed."

Any chance of getting scopy to run on this device ?

ioanachelaru, 2022-03-21:
Hi,

At the moment, Scopy requires at least API 28, which is equivalent to Android 9 (Pie). Will be closing this for now.

adisuciu, 2022-03-24:
Hi, 

I spent a little time investigating the build, and I think I've gotten the Android version requirements down to API level 26 which is Android 8.0. I tested it locally and it seems to work although I'm on android 10 .. Would you mind giving this a test ? If all is working fine, I will update the build steps.

(github does not support attaching apk files or zip files that large in comments so i uploaded it to transfer.sh - link will probably expire in a week or two) - https://transfer.sh/fw3QEA/android-build-debug.apk

jpnielsen, 2022-03-24:
Thank you for looking into this:

rk3288_10P:/storage/emulated/0/Download # pm install android-build-debug.apk
Failure [INSTALL_FAILED_NO_MATCHING_ABIS: Failed to extract native libraries, res=-113]

rk3288_10P:/storage/emulated/0/Download # uname -a
Linux localhost 4.4.132 #2 SMP PREEMPT Thu Dec 17 11:50:04 CST 2020 armv7l

` 
cat /system/build.prop

# begin build properties
# autogenerated by buildinfo.sh
ro.build.id=OC
ro.build.display.id=APPC-15X.AP6255_NL.PRODVX.EDP-1080P.V1.1_1.11.11c.20201228
ro.build.version.incremental=APPC-15X.AP6255_NL.PRODVX.EDP-1080P.V1.1_1.11.11c.20201228
ro.build.version.sdk=27
ro.build.version.preview_sdk=0
ro.build.version.codename=REL
ro.build.version.all_codenames=REL
ro.build.version.release=8.1.0
ro.build.version.security_patch=2020-03-05
ro.build.version.base_os=
ro.build.date=Mon Dec 28 16:15:28 CST 2020
ro.build.date.utc=1609143328
ro.build.type=userdebug
ro.build.user=roper
ro.build.host=cs1
ro.build.tags=test-keys
ro.build.flavor=rk3288_10P-userdebug
ro.product.model=APPC-15X
ro.product.brand=PRODVX
ro.product.name=rk3288_10P
ro.product.device=rk3288_10P
# ro.product.cpu.abi and ro.product.cpu.abi2 are obsolete,
# use ro.product.cpu.abilist instead.
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
ro.product.cpu.abilist=armeabi-v7a,armeabi
ro.product.cpu.abilist32=armeabi-v7a,armeabi
ro.product.cpu.abilist64=
ro.product.manufacturer=ZDMID
ro.product.locale=en-US
ro.wifi.channels=
# ro.build.product is obsolete; use ro.product.device
ro.build.product=rk3288_10P
# Do not try to parse description, fingerprint, or thumbprint
ro.build.description=rk3288_10P-userdebug 8.1.0 OC APPC-15X.20201228 test-keys
ro.build.fingerprint=PRODVX/rk3288_10P/rk3288_10P:8.1.0/OC/roper12281615:userdebug/test-keys
ro.build.characteristics=tablet
# end build properties
#
# from device/rockchip/rk3288_10P/system.prop
#
#
# system.prop
#

#rild.libpath=/system/lib/libreference-ril.so
#rild.libargs=-d /dev/ttyUSB2
# Default ecclist
ro.ril.ecclist=112,911
ro.opengles.version = 131072
wifi.interface=wlan0
rild.libpath=/system/lib/libril-rk29-dataonly.so
rild.libargs=-d /dev/ttyACM0
persist.tegra.nvmmlite = 1
ro.audio.monitorOrientation=true

#NFC
debug.nfc.fw_download=false
debug.nfc.se=false

#add Rockchip properties here
ro.rk.screenoff_time=0x7fffffff
ro.rk.screenshot_enable=true
ro.rk.def_brightness=200
ro.rk.homepage_base=http://www.google.com/webhp?client={CID}&amp;source=android-home
ro.rk.install_non_market_apps=true
sys.hwc.compose_policy=6
sys.wallpaper.rgb565=0
sf.power.control=2073600
sys.rkadb.root=0
ro.sf.fakerotation=false
#ro.sf.hwrotation=0
ro.rk.MassStorage=false
ro.rk.systembar.voiceicon=true
ro.rk.systembar.tabletUI=false
ro.rk.LowBatteryBrightness=false
wifi.interface=wlan0
ro.tether.denied=false
sys.resolution.changed=false
ro.default.size=100
persist.sys.timezone=Asia/Taipei
ro.product.usbfactory=rockchip_usb
wifi.supplicant_scan_interval=15
ro.factory.tool=0
ro.kernel.android.checkjni=0
#set default lcd density to Rockchip tablet
ro.sf.lcd_density=160
ro.adb.secure =1
#force camera API 1
camera2.portability.force_api=1
#set wifi contry code
#ro.boot.wificountrycode=CN

# manual rotation default value.
#persist.sys.rotation=0
#persist.demo.hdmirotates=true

#systemUI add power off
#ro.rk.systembar.powerofficon=true

#service.adb.tcp.port=5555
persist.sys.EnableProxy=true

# Keypad ctrl
persist.sys.adc = enabled

# Touch ctrl
persist.sys.touchenable = true
persist.sys.touchwake = true

# USB ctrl
persist.sys.usbenable = true
#
# ADDITIONAL_BUILD_PROPERTIES
#
ro.bionic.ld.warning=1
ro.treble.enabled=true
persist.sys.dalvik.vm.lib.2=libart.so
dalvik.vm.isa.arm.variant=cortex-a15
dalvik.vm.isa.arm.features=default
dalvik.vm.lockprof.threshold=500
net.bt.name=Android
dalvik.vm.stack-trace-dir=/data/anr
ro.expect.recovery_id=0x6b5b8eb34fda8fe9496bf2d8c7662734cd540353000000000000000000000000
`

adisuciu, 2022-03-24:
Okay, I think that the processor you are running is 32bit
```
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
ro.product.cpu.abilist=armeabi-v7a,armeabi
ro.product.cpu.abilist32=armeabi-v7a,armeabi
ro.product.cpu.abilist64=
```

The build is for 64 bit only (aarch64). I kind of deprecated the 32bit build somewhere along the development process. I'll check if it still builds ..

adisuciu, 2022-03-28:
Can you try this - https://transfer.sh/b9HJus/android-build-debug.apk please ? I think I nailed everything down ..

jpnielsen, 2022-03-28:
Thank you.

It runs, but not very fast - I will use Scopy on Windows 11 and Linux in the future.

adisuciu, 2022-03-28:
Cool, just wanted to make sure it starts as I don't own a 32bit tablet with Android 8. I'll update the builds .. Thanks !

closed 2022-03-28 by adisuciu