analogdevicesinc/scopy #1238: Pattern Generator: Preview plot does not automatically change unless Run is clicked

tagoylo, 2022-03-25:
**Environment:**
 - OS: macOS Catalina 10.15.7
 - FW Version: 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
Preview plot does not automatically change unless Run is clicked. It is observed when channels are in group. It also appears in individual channel, especially when Number pattern is selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Pattern Generator instrument.
2. Enable 2 channels and create a group.
3. Change pattern to clock.
4. The preview does not change immediately. Unlike when in individual.
5. Click Run. The preview plot will now show the expected plot.

**Expected behavior**
Preview automatically updates.

**Screenshots**
[image]

IonutMuthi, 2022-03-25:
this should fix the issue https://github.com/analogdevicesinc/scopy/pull/1243

tagoylo, 2022-04-27:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-27 by tagoylo