analogdevicesinc/scopy #1249: Simple mode for smaller screen size

tagoylo, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
I've been using my phone (6.7in screen size) with scopy since I don’t have a tablet with me.
It is working and all important features are accessible. But one thing to note is that most of the parts are small (e.g. buttons, plots like histogram in oscilloscope). Sometimes it takes time to tap the right area. Ex. When All channels are enabled in oscilloscope plus the reference/math channels, the area to select is very tight and difficult to navigate. Maybe, this is not an issue if the screen size is big, like a tablet. See picture:

[image]
[image]

**Describe the solution you'd like**
Maybe having a simple mode for smaller screen sizes android devices? Like all important features are there but the UI is more small screen size friendly. Maybe add to preferences an option to switch to simple mode.

adisuciu, 2022-03-28:
The application is really targeted towards tablets.
There is a preferences option to do tabbed plots in the oscilloscope which may help with the clutter. There is not much that we can do right now regarding number of channels.

tagoylo, 2022-03-28:
This is noted. Should I close this item or keep it open?

adisuciu, 2022-03-28:
Close for now

closed 2022-03-28 by adisuciu