analogdevicesinc/scopy #1250: Logic Analyzer: Capture panel (graph) not automatically updated if triggered.

kimpaller, 2022-03-28:
**Environment:**
 - OS: Windows
 - FW Version: 0.28
 - SW Version or git tag: v1.4.0 rc1 https://github.com/analogdevicesinc/scopy/commit/895c0e98615a4579dca1746db63b061db25ef18c

**Describe the bug**
Need to click "Stop" first before graph is loaded on capture panel.  Observed in triggers - '**Rising Edge**', '**Falling Edge**' and '**Any Edge**'

**To Reproduce**
Steps to reproduce the behavior:
1. Enable channel DIO0 and DIO1 on Logic analyzer, on the settings menu set sample rate to 50ksps and add a delay ex. -142 samples. On Pattern generator, enable DIO1, set the following parameter: Pattern: clock, Frequency: 100Hz, Phase: 0 degrees and Duty Cycle: 50%. On Digital IO instrument, set DIO0 as output.
2. Set DIO0’s trigger to rising edge configuration, run Digital IO, pattern generator and logic analyzer instrument. Change DIO0’s output from 0 to 1 from the Digital IO’s instrument.
3.  The Logic Analyzer should initiate a capture on the rising edge of DI0 and update graph correspondingly. However, graph is not updated unless instrument is stopped.

**Expected behavior**
The logic analyzer should initiate a capture on the rising edge and automatically update graph on the capture panel without stopping the instrument.

**Screenshots**

Expected behavior: on trigger, capture graph should be updated without stopping instrument like the example below:
[image]

However, the behavior is not observed, and graph is only updated when "Stop" is clicked.

[image]

**Additional context**

The following .ini can be used for replicating the issue

[logicanalyzer_failed_capture_update.zip](https://github.com/analogdevicesinc/scopy/files/8359637/logicanalyzer_failed_capture_update.zip)

adisuciu, 2022-03-29:
I cannot reproduce this behavior on my machine. Everything works as expected .. Can you still reproduce using the provided ini ? Can you reproduce this consistently ?

kimpaller, 2022-03-30:
After machine reboot and clearing out C:\Users\your_username\AppData\Roaming\ADI , the issue is no longer reproducible with same .ini used. Probably just an isolated case. Will close for being irreproducible.

closed 2022-03-30 by kimpaller