analogdevicesinc/scopy #1257: Voltmeter: "Out of Range" warning if input is -2.5V to 0V in +/-2.5V Range selected

tagoylo, 2022-03-29:
**Environment:**
 - OS: linux-arm32, windows10
 - FW Version: 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
here is always a "out of range" warning when range is set to +/-2.5V and the input voltage is negative. Ex. a -1V from negative supply would still enable the warning though it is within range. This happens on both channels.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect negative supply pin to Ch1 or Ch2 analog in.
2. Open power supply. Set negative supply to -1V.
3. Open Voltmeter, set to DC. Set Range to +/-2.5V.
4. Run both instrument.
5. Monitor voltmeter. Notice that there is a "Out of range" status/ warning even though input is -1V and is within the +/-2.5 V range.
6. Change Range to +/-25V and the status is gone.

**Expected behavior**
Show warning only if input is out of range.

**Screenshots**
[image]

adisuciu, 2022-03-29:
https://github.com/analogdevicesinc/scopy/pull/1265

kimpaller, 2022-03-31:
Addition observation(s):

If voltmeter is set to high gain mode ( +-25V) for low voltage inputs (e.g ~100mV, ~-100mV ), it is expected that "Out of range" warning be displayed like the ff. screen capture:

[image]

However no warnings was observed for the current version:

[image]

adisuciu, 2022-03-31:
I do not agree, I think 0V is within [-25, 25V] therefore no message should be displayed.

tagoylo, 2022-04-28:
I tried +-2.4 V as input but the message is still displayed. Is this expected?
[image]

adisuciu, 2022-04-28:
(https://github.com/analogdevicesinc/scopy/pull/1333)

closed 2022-05-17 by adisuciu