analogdevicesinc/scopy #1259: General: Widgets windows are somewhat transparent in linux

tagoylo, 2022-03-29:
**Environment:**
 - OS: linux-arm32
 - FW Version: 0.28 
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
Widgets windows are somewhat transparent in the sides in linux-arm32. It is not the case in windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Preferences.
2. Enable dockable widgets.
3. Go to oscilloscope or any instruments with plot.
4. Click on the ">" to detach widget.
5. Notice that the widget window is transparent on the sides.

**Screenshots**
[image]

adisuciu, 2022-03-29:
That's weird it does not seem to reproduce on Kuiper raspberry-pi (arm) nor on my Ubuntu PC ..

tagoylo, 2022-03-29:
I will confirm this on another Linux machine.

adisuciu, 2022-03-30:
I think this may be related to Ubuntu 21.10 but cannot test. Do you have it running on a PC(not raspi) ?

adisuciu, 2022-03-30:
Can you run Scopy like this ?  XDG_SESSION_TYPE=x11 flatpak run org.adi.Scopy

tagoylo, 2022-03-31:
Running Scopy with command `XDG_SESSION_TYPE=x11 flatpak run org.adi.Scopy` solved the issue in Ubuntu 21.10 on Rpi4. Widgets are now not transparent. See screenshot below.
[image]

tagoylo, 2022-05-02:
Using the command _XDG_SESSION_TYPE=x11 flatpak run org.adi.Scopy_ is also applicable in rc2. Closing this item.

closed 2022-05-02 by tagoylo