analogdevicesinc/scopy #1260: General: Closed widget does not return to main instrument window in linux

tagoylo, 2022-03-29:
**Environment:**
 - OS: linux-arm32
 - FW Version : 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
Closed widget does not return to main instrument window. Also, dragging the widget into the main instrument window does not return/attach it again. This is not the case in windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Preferences.
2. Enable dockable widgets.
3. Go to oscilloscope or any instruments with plot.
4. Click on the ">" to detach widget.
5. Close widget.
6. Notice that it does not return to the instrument window.

Repeat 1-4
5. Drag widget to main instrument window.
6. Notice that it does not return to the instrument window.

**Screenshots**
[image]

adisuciu, 2022-03-29:
On kuiper there are more buttons than expected .. 
[image]

Red will close the widget completely. Yellow should return it back to the dock .. Also dragging it by purple should allow redocking. Blue shouldn't. 

This is not the case on ubuntu: 
[image]
Red from previous image does not exist here.

Can you confirm that the behavior is as described above. Also can you confirm what the detached plot looks like ?
-Adrian

tagoylo, 2022-03-30:
For Ubuntu 21.1 in rpi4 (linux-arm32 scopy version), this is what the detached plot looks like.
[image]
The controls, there is no downward arrow at the upper right for redocking. An "x" also appears at the upper right which closes entirely the detached plot and not redocking it.

For Rpi OS (not Kuiper)  (linux-arm32 scopy version), this is what the detached plot looks like.
[image]
The behavior is the same as what you've explained for Kuiper except for the "x" on the upper right which is disabled in this case.

Will update this comment for Kuiper.

adisuciu, 2022-03-30:
Thanks for your insight. I think this is related to Ubuntu 21.10. We don't currently have a machine that does that. If you can confirm on a PC (not raspi) it would be great. If you don't have it, we can wait for 22.04 which is a LTS and we will do a fix then.

tagoylo, 2022-05-02:
Using the command XDG_SESSION_TYPE=x11 flatpak run org.adi.Scopy is also applicable in rc2. Closing this item.