analogdevicesinc/scopy #1261: Voltmeter: Label "Notes" in bigger font

tagoylo, 2022-03-29:
**Environment:**
 - OS: linux-arm32, windows10
 - FW Version : 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
When notes per instrument is enabled, the "Notes" label in voltmeter is in bigger font compared to what is in other instruments.

**To Reproduce**
Steps to reproduce the behavior:
	1. Open Preferences.
	2. Enable all instrument notes.
	3. Go to voltmeter.
	4. Notice that the "Notes" label is in bigger font.

**Expected behavior**
Should be the same with other instruments.

**Screenshots**
[image]

adisuciu, 2022-03-29:
https://github.com/analogdevicesinc/scopy/pull/1262

tagoylo, 2022-04-27:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-27 by tagoylo