analogdevicesinc/scopy #1268: Oscilloscope: Histogram vertical axis is compressed even at increased Scopy window size

jpineda3, 2022-03-30:
**Environment:**
 - OS: Windows 10, macOS 11.4 Big Sur
 - FW Version: v0.28
 - SW Version or git tag: v1.4.0 rc1 https://github.com/analogdevicesinc/scopy/commit/895c0e98615a4579dca1746db63b061db25ef18c

**Describe the bug**
The vertical axis appears compressed in the histogram. The counts/percentage labels overlap and are unreadable.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Scope Ch1 to AWG Ch1.
2. Go to Signal Generator and generate any type of signal. For the example below a 3Vpp square wave was generated.
3. Under Oscilloscope general settings, turn on Histogram.
4. Run both Signal Generator and Oscilloscope.
5. See error

**Expected behavior**
The vertical axis labels should be spaced evenly and readable as in the previous release.
[image]

**Screenshots**
This is the error.
[image]

**Additional context**
[histogram_vertical.zip](https://github.com/analogdevicesinc/scopy/files/8375898/histogram_vertical.zip)

AlexandraTrifan, 2022-04-01:
Fixed with PR #1295

jpineda3, 2022-04-28:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-28 by jpineda3