analogdevicesinc/scopy #1269: Oscilloscope: Cannot drag widget window to left when histogram is enabled

tagoylo, 2022-03-30:
**Environment:**
 - OS: linux-arm32 (rpi OS)
 - FW Version : 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
When all oscilloscope views are enabled, especially Histogram, user cannot expand the windows (FFT/XY) to the left.

**To Reproduce**
Steps to reproduce the behavior:
	1. Open Preferences.
	2. Enable dockable widgets.
	3. Got to Oscilloscope.
	4. Enable the following: FFT, XY, and Histogram.
	5. Use the dotted line in the center to drag widgets window to resize.
	6. Try dragging it left and right.
	7. Notice that, when dragging it left, it does not fully expand to left. Unlike when dragging to right.
	8. Disable Histogram.
        9. Try dragging left again, now it expands further.

**Screenshots**
[image]

AlexandraTrifan, 2022-04-01:
Fixed with PR #1295

tagoylo, 2022-04-27:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-27 by tagoylo