analogdevicesinc/scopy #1270: General: No label on other detached plot

tagoylo, 2022-03-30:
**Environment:**
 - OS: linux-arm32 (rpiOS)
 - FW Version :0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
No label on other detached plots. For oscilloscope views, there is a labels such as Time Domain, FFT, XY, Histogram. But for other plots, it is only "Scopy". If there are many detached plots, there's confusion as to which is which.

**To Reproduce**
Steps to reproduce the behavior:
	1. Open Preferences.
	2. Enable dockable widgets.
	3. Detach plots e.g. in oscilloscope, network analyzer, signal generator, etc.
	4. Minimize these windows.
	5. Notice that the labels of the plots are "Scopy", there's no way to know whose plot is that window unless user opens it one by one.

**Expected behavior**
Add specific label to detached plots.

**Screenshots**
[image]

AlexandraTrifan, 2022-04-01:
Fixed with PR #1295

tagoylo, 2022-04-29:
Labels are verified implemented in v1.4. rc2. Observed that only network analyzer and oscilloscope have the dockable widgets feature, is this expected?

adisuciu, 2022-04-29:
Yes, as they are the onlyones that have multiple plots

closed 2022-04-29 by adisuciu