analogdevicesinc/scopy #1271: Oscilloscope: Right arrow for tabbed plots does not work

tagoylo, 2022-03-30:
**Environment:**
 - OS: linux-arm32 (rpiOS), Windows10
 - FW Version : 0.28
 - SW Version or git tag: v1.4.0 rc1 895c0e9

**Describe the bug**
The right arrow for tabbed plots for oscilloscope views does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Preferences.
2. Enable dockable widgets.
3. Go to Oscilloscope.
4. Enable all views. Select XY tab.
5. Expand left plots to the right. (See picture)
6. Tabs (FFT and XY) are compressed and "<" and ">" arrows are visible.
7. Click on "<" to view FFT tab.
Try clicking ">" to view again XY tab but it does not respond.

**Expected behavior**
The XY tab should show after clicking ">". 

**Screenshots**
[image]

adisuciu, 2022-03-30:
This does not reproduce for me, the arrows work fine. Also this happens really due to the histogram bug where you cannot resize the plot. We're investigating that.

AlexandraTrifan, 2022-04-01:
I was able to reproduce it on MacOS, when the FFT/XY tab is resized to its minimum width. 
I added some changes that fix the arrow behaviour in PR #1295

tagoylo, 2022-04-27:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-27 by tagoylo