analogdevicesinc/scopy #1275: ADALM2000: Logic analyzer: Show table of decoded messages

frmdstryr, 2022-03-30:
**Is your feature request related to a problem? Please describe.**

The logic analyzer decoding works great however I can only see one packet at a time...

**Describe the solution you'd like**

Show a table of all decoded messages.

**Additional context**

Zoomed out:
[image]

Zoomed in:

[image]

adisuciu, 2022-03-31:
This would indeed be cool. There currently is an alternative where you could use sigrok-cli in conjunction with the m2k-cli to get a list of decoded messages. You wouldn't be able to visualize the data however. https://wiki.analog.com/university/tools/m2k/libm2k/m2kcli?s[]=m2kcli#m2kcli_and_sigrok-cli

frmdstryr, 2022-03-31:
It looks pulseview added something like this recently: https://www.sigrok.org/blog/new-pv-tabular-decoder-output-view

For now I'll see if I can get something using the cli and hopefully view it with that, thank you!

frmdstryr, 2022-04-13:
I managed to get something working in scopy.  It currently only works for usb packets because that's I need it for at the moment. 

[image]

I can't figure out how to make it sync with the view. Eg when clicking a packet in the table. I'd like to have the plot zoom into the range of  min sample + max sample. Is there a function for that somewhere? 

See https://github.com/frmdstryr/scopy/tree/decoder-table

As a side note it would probably be a good idea to run all the code through some kind of style formatter and fix some of the compile warnings. I fixed a couple obvious ones but there is still a lot..

adisuciu, 2022-04-14:
Hi, 

This is amazing work ! 

What you are looking for is the zoom function .. I created this commit on a mirror of your fork here - https://github.com/analogdevicesinc/scopy/tree/frmdstryr-decoder-table

The style and warning analysis are implemented in CI in a pull request -  we're just waiting to release and then we merge it - https://github.com/analogdevicesinc/scopy/pull/1205

-Adrian

frmdstryr, 2022-04-14:
Ah, that's perfect, thanks! After that comment I got it to center and I'm unsure if zooming is the best approach because it hides the packets around. Maybe double click could do center + zoom or just do a zoom + padding or something. 

Can you think of a better way to split the capture into table rows? It is currently limited to one decoder row which doesn't work well with certain decoders (eg uart has rx and tx packets). I might change it to take a set of decoder row indexes unless you have any ideas. 

Also it would be nice to be able to filter out unwanted stuff (eg usb SOF packets) and optionally certain rows (the bitfields are a bit excessive) but I'm not sure where that should go in the interface.

It'd be nice if the side panel view could be resized but I didn't want to change too much (would need to change the whole layout to use a QSplitter I think).

frmdstryr, 2022-04-14:
I like the search idea. It the search could almost be at the top next to the print/group buttons? It wouldn't really need the decode table to search.

Since space is limited I don't like splitting up the table area that much... I think it would make more sense to go somewhere in the channel settings somewhere with the decoder configuration or have a dropdown somewhere in the decode side panel.

adisuciu, 2022-04-14:
> Yeah that would be nice. I didn't see any click signals but didn't dig around that much. Worst case it could manually check the click area against painted bounds or something.

Selecting the channel/decoder(by double clicking the handle) already updates channel settings. It's a question of also updating the decoder tab. I think this method is called when a channel handle is selected on the left. https://github.com/analogdevicesinc/scopy/blob/master/src/logicanalyzer/logic_analyzer.cpp#L1205. The signal is sent from the plot.

> I like the search idea. It the search could almost be at the top next to the print/group buttons? It wouldn't really need the decode table to search.

Since we can only really search for decoded annotations, I think it makes sense having it in the menus. Maybe the string should read "Filter " instead of search.

> Since space is limited I don't like splitting up the table area that much... I think it would make more sense to go somewhere in the channel settings somewhere with the decoder configuration or have a dropdown somewhere in the decode side panel.

It makes sense in the decoder configuration only if it filters out the rows on the plot as well (which is not necessarily bad). If not it should be in the decoder tab imo. If it's an issue of space, tabs or collapsible widgets can be created such as the ones in the signal generator.

[image]

[image]

I think this could turn into a really useful feature !

frmdstryr, 2022-04-15:
I pushed few more commits that scrolls the table to the decoded message when clicking the "primary" decoder row on the plot. Also adds a way to toggle which decoder rows are displayed but is not hooked up to any interface. It should highlight what's selected but it's better than before.

The zoom method seemed to flicker and messed up the zoom stack (when right clicking) so I made it go back to centering for now.

The collapsible seems like a good way too do it.

adisuciu, 2022-04-20:
What do you mean by flicker/messed up the zoom stack ? 
z->zoom(0) should go back to the zoom base, and z->zoom(r) should create a new level in the stack. Therefore, when clicking on a decoder, the stack should have 2 levels (base + decoder level). Rightclicking should simply return to the base.  Isn't this the behavior you were seeing ?

-Adrian

frmdstryr, 2022-04-20:
Yes that's what it was doing. By the flicker I mean if you click one packet in the table, then click another another right next to it the plot zooms out then back in when it could just move over.  

This program https://www.youtube.com/watch?v=Ak9R4yxQPhs supports both methods (scrolling the horizontal position and scroll+zoom to packet). So that would be an option. I also like the text view in that program and the hovering to show timing is neat.

Rendering the decoder rows in the table might not be the best approach, not sure what you're thoughts are on that?

adisuciu, 2022-04-20:
The flickering of the zoom could probably be implemented using some logic like if zoomed to packet, do not reset the stack but rather overwrite this zoom level.

I think that scrolling the horizontal position could be implemented by doing a middle mouse button to click and drag the plot ( by doing this action you essentially do the same thing as dragging the buffer previewer at the top).  

The hover to show timings is neat indeed. In terms of implementation it requires setting a tracker mode to the zoomer and overriding the QwtPicker implementation attached to the zoomer to show the correct timings when hovering. It would also need a way to plot the "pulse delimiters". This may prove to be quite challenging and I think it would be the subject of a different PR.

The text mode table could be useful for people that want to export decoder data and want to do some parsing programatically. sigrok-cli is very flexible in this regard, but i guess it would be a nice-to-have feature to do something like this out of the box. In terms of rendering decoder rows looks nice, but having some additional metadata such as start-offset and decoded message width would be even more useful in my opinion. I don't know if there's a way to show both decoder rows and the metadata ...

-Adrian

adisuciu, 2022-07-04:
We will continue working on this code and implement the features we discussed here if that's okay with you.
-Adrian

frmdstryr, 2022-07-04:
That would be great! I appreciate all of your help.

adisuciu, 2022-09-02:
https://github.com/analogdevicesinc/scopy/pull/1370

frmdstryr, 2022-09-02:
Wow that looks great! Awesome work! I should be able to give it a try next week.

I was using the original table a last week and one other thing I thought would be helpful (at least for usb) would be the ability to copy the decoded message to the clipboard.

closed 2024-09-24 by adisuciu