analogdevicesinc/scopy #1279: Oscilloscope: Mixed Signal View text does not fit in button

jpineda3, 2022-03-31:
**Environment:**
 - OS: Windows 10
 - FW Version: v0.28
 - SW Version or git tag: v1.4.0 rc1 https://github.com/analogdevicesinc/scopy/commit/895c0e98615a4579dca1746db63b061db25ef18c 

**Describe the bug**
The "d" in Enable/Disable Mixed Signal view is not fully displayed in the button. At least on my Windows, the issue is only during Disable and the "d" is partially displayed. On others', the d is completely gone.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Oscilloscope.
2. Check if the complete text is visible in the "Enable Mixed Signal View" button.
3. Check if the complete text is visible in the "Disable Mixed Signal View" button.

**Expected behavior**
The full text should be visible.

**Screenshots**
[image]
[image]

**Additional context**
[mixed_signal.zip](https://github.com/analogdevicesinc/scopy/files/8385553/mixed_signal.zip)

AlexandraTrifan, 2022-04-05:
Fixed with PR #1310

jpineda3, 2022-04-28:
Bug verified fixed in v1.4.0 rc2. Closing this item.

closed 2022-04-28 by jpineda3