analogdevicesinc/scopy #1282: General: Detached plots in windows has no tab when minimized

tagoylo, 2022-03-31:
**Environment:**
 - OS: Windows 10
 - FW Version : 0.28
 - SW Version or git tag: v.1.4.0 rc1 895c0e9

**Describe the bug**
When minimizing detached plot window, there is no individual tab that can be clicked again to reopen the minimized tab.  Unless main scopy window is closed and opened again, all the detached plot windows reopens too. This only happens on Windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Preferences.
2. Enable dockable widgets.
3. Detach plots e.g. in oscilloscope, network analyzer, signal generator, etc.
4. Minimize these windows.
5. For Windows 10, there is no minimized tab that can be clicked again to reopen. Unless main scopy window is closed and opened again, the detached plot windows reopens too.

**Expected behavior**
There's a tab for minimized window.

**Screenshots**
[image]

ioanachelaru, 2022-03-31:
Hi,

The expected behavior on Windows is that if the docked window is minimized, you can recover it by minimizing and maximizing again the main window. Can you confirm that this is the case?

-Ioana

tagoylo, 2022-04-01:
Hi,
Yes, the docked widget window can be recovered it by minimizing and maximizing again the main window.

adisuciu, 2022-04-04:
This is what the library provides for windows. Unfortunately there is no easy fix for this that I'm aware of. Closing.