analogdevicesinc/scopy #1284: Oscilloscope: Mixed signal view, decoder added by scrolling down in "select a decoder to add" area

tagoylo, 2022-03-31:
**Environment:**
 - OS: linux-arm32(ubuntu21.10, rpiOS), windows 10
 - FW Version : 0.28
 - SW Version or git tag: v1.4.0 rc1 rc1 895c0e9

**Describe the bug**
In Mixed signal view, a decoder is added automatically just by scrolling down in "select a decoder to add" area. In logic analyzer, this does not happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Oscilloscope.
2. Enable Mixed Signal View.
3. Move cursor to "Select a decoder to add" area.
4. Scroll down using mouse's scroll.
5. Notice that a decoder is automatically added. In linux, it is decoder ac97. In Windows, it is ir_rc5.

**Expected behavior**
No decoder added automatically.

**Screenshots**
[image]

IonutMuthi, 2022-03-31:
this should fix the issue https://github.com/analogdevicesinc/scopy/pull/1292

tagoylo, 2022-04-27:
Upon initial opening of mixed signal view and scrolling through the area, the issue does not reproduce anymore. But it shows after selecting a decoder.

Steps to reproduce:
1. Go to Oscilloscope.
2. Enable Mixed Signal View.
3. Move the cursor to "Select a decoder to add" area.
4. Scroll down using the mouse's scroll. At this point, the issue is not reproducible anymore.
5. Click "Select a decoder to add".
6. Select any decoder or just close selection without choosing.
7. Scroll down again on the area and the issue reappears.
8. Delete the added decoders and then scroll again on the area, it is observed that the issue is again not reproducible.

adisuciu, 2022-04-28:
Fixed with https://github.com/analogdevicesinc/scopy/pull/1329

closed 2022-05-17 by adisuciu