analogdevicesinc/scopy #1336: General: Scopy crashes when called on the command line  with --script <js script> option (observed on Windows)

kimpaller, 2022-05-02:
**Environment:**
 - OS: Windows 10
 - FW Version: 0.28 
 - SW Version or git tag: [v1.4.0 rc2]( https://github.com/analogdevicesinc/scopy/commit/cca49b2ba901f8d8c40327c689761c87db368fa9)

**Describe the bug**
Scopy crashes when called on the command line  with --script option. This was observed however on windows machines only.

**To Reproduce**
Steps to reproduce the behavior:
1. Add C:\Program Files\Analog Devices\Scopy to environment variables
2. Create a scopy (js) script or get one from https://github.com/analogdevicesinc/scopy/tree/master/js
3. Go to the directory where the js script is located.
4. Execute: Scopy.exe --script script_name.js

**Expected behavior**
Scopy should open and execute script.

kimpaller, 2022-05-05:
Hi Alexandra, I've tried the startupscript.js under https://github.com/analogdevicesinc/scopy/tree/master/js., just changed line 4 to var host = "192.168.2.1".  No error message was displayed, the application window attempts to open but crashed immediately.

This was the command i've used:

`Scopy.exe --script startupscript.js`

AlexandraTrifan, 2022-05-05:
Hi,

I tested the "startupscript.js" script again and it seems to work. Scopy starts up, I get 2 pop-ups (one regarding license and one regarding preferences), osc and sig gen are running and I then have full control of the application. 

Can you clean up all the .ini files (Scopy.ini, Scopy.ini.bak and Preferences.ini) and try to run the script again?

-Alexandra

kimpaller, 2022-05-10:
Hi Alexandra,

I've tried cleaning all the .ini files and the script was executed as expected. Will close this issue now as it is no longer reproducible with the procedure provided. Thanks!

closed 2022-05-10 by kimpaller